danjulio/lepton #1: Troubles in teensy3 code

Konan23, 2020-09-05:
Hi @danjulio , 

I have some troubles in your teensy3 code. I tried to verify lep_test5 and had errors 
`lep_test5:265: error: 'IRQ_PORTA' was not declared in this scope
   SPI.notUsingInterrupt(IRQ_PORTA);
`
`controls:85: error: 'INTERNAL' was not declared in this scope
   analogReference(INTERNAL);
`
I couldn't find IRQ_PORTA and INTERNAL variables was declared or have instruction about them. Could you solve it pls. Thank you!

danjulio, 2020-09-05:
Hello @Konan23 ,

INTERNAL and IRQ_PORTA are part of the PJRC Teensy addition to the Arduino environment.  Have you included support for the Teensy in your Arduino application?  And have you selected "Teensy 3.2/3.1" as your board type?

Konan23, 2020-09-09:
Thanks for your reply and after trying to fix something, I still have error about multiple definition. I searched that error come from the overlap between 2 library i2c_t3 and Wire. I tried to deactivate Wire library in source code and header file but it doesn't work. Here is my error message 

> Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"
C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o: In function 'i2c0_isr':

> C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\Wire/WireKinetis.cpp:912: multiple definition of 'i2c0_isr'

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.cpp:1134: first defined here

> c:/program files (x86)/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o: In function 'Print::availableForWrite()':

> C:\Program` Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\Wire/WireKinetis.h:136: multiple definition of 'i2c1_isr'

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:883: first defined here

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o: In function 'Print::availableForWrite()':

> C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\Wire/WireKinetis.h:136: multiple definition of 'Wire1'

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:883: first defined here

> c:/program files (x86)/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol 'Wire1' changed from 20 in 

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o to 108 in C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o: In function 'Print::availableForWrite()':

> C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\Wire/WireKinetis.h:136: multiple definition of 'Wire'

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o:C:\Program Files (x86)\arduino-1.8.13\hardware\teensy\avr\libraries\i2c_t3/i2c_t3.h:883: first defined here

> c:/program files (x86)/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol 'Wire' changed from 20 in 

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\i2c_t3\i2c_t3.cpp.o to 108 in 

> C:\Users\alta\AppData\Local\Temp\arduino_build_572620\libraries\Wire\WireKinetis.cpp.o

> collect2.exe: error: ld returned 1 exit status

> Error compiling for board Teensy 3.2 / 3.1.

Could you give me some advice to fix them. Thank you so much!

danjulio, 2020-09-11:
I'm sorry, I'm not sure I have much to offer you.  My LeptonSDKEmb32OEM.cpp file includes i2c_t3.  Maybe something else is including either i2c_t3 or wire and there is a collision.  All I can suggest is to look at every file and see if you can find a way multiple files are being included.

Konan23, 2020-09-12:
Thank you for your help. I will try some cases and hope that it will work. I will give you feedback when I completed it.

Konan23, 2020-09-17:
Finally, I debugged it successfully. The new version of Adafruit_GFX_Library contains Adafruit_GrayOLED file and it uses Wire.h to communication so when I compile the code, the program will call it too. Just delete the header and source files of Adafruit_GrayOLED, the program will run. 

@danjulio you can close this issue :)))

danjulio, 2020-09-17:
Glad to hear you fixed it!

closed 2020-09-17 by danjulio