danjulio/lepton #3: idf.build error

JungSuWhan, 2021-04-06:
Hello Thank you for sharing a great project.

I ran a build with the idf.build command, but I got a compile error, so I'm inquiring.

../components/cmd/json_utilities.h:36:10: fatal error: ds3232.h: No such file or directory
 #include "ds3232.h"
          ^~~~~~~~~~
compilation terminated.
[7/24] cd /home/jsh/working/lepton/ESP32/tCam-Mini/firmware/build/esp-idf/part...*****************************************************************************
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[7/24] Building C object esp-idf/lepton/CMakeFiles/__idf_lepton.dir/cci.c.obj
FAILED: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I../components/lepton -I../main -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/soc/esp32/include -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/xtensa/include -I/root/esp/esp-idf/components/xtensa/esp32/include -I/root/esp/esp-idf/components/esp32/include -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/esp_ringbuf/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/tcpip_adapter/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/esp32/include -I/root/esp/esp-idf/components/lwip/port/esp32/include/arch -I/root/esp/esp-idf/components/vfs/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_wifi/esp32/include -I/root/esp/esp-idf/components/esp_eth/include -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32/include -I/root/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.2\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/lepton/CMakeFiles/__idf_lepton.dir/cci.c.obj -MF esp-idf/lepton/CMakeFiles/__idf_lepton.dir/cci.c.obj.d -o esp-idf/lepton/CMakeFiles/__idf_lepton.dir/cci.c.obj   -c ../components/lepton/cci.c
../components/lepton/cci.c:25:10: fatal error: i2c.h: No such file or directory
 #include "i2c.h"
          ^~~~~~~
compilation terminated.
[7/24] Building C object esp-idf/lepton/CMakeFiles/__idf_lepton.dir/vospi.c.obj
FAILED: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I../components/lepton -I../main -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/soc/esp32/include -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/xtensa/include -I/root/esp/esp-idf/components/xtensa/esp32/include -I/root/esp/esp-idf/components/esp32/include -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/esp_ringbuf/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/tcpip_adapter/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/esp32/include -I/root/esp/esp-idf/components/lwip/port/esp32/include/arch -I/root/esp/esp-idf/components/vfs/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_wifi/esp32/include -I/root/esp/esp-idf/components/esp_eth/include -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32/include -I/root/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.2\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/lepton/CMakeFiles/__idf_lepton.dir/vospi.c.obj -MF esp-idf/lepton/CMakeFiles/__idf_lepton.dir/vospi.c.obj.d -o esp-idf/lepton/CMakeFiles/__idf_lepton.dir/vospi.c.obj   -c ../components/lepton/vospi.c
In file included from ../components/lepton/vospi.c:38:
../components/lepton/vospi.h:31:10: fatal error: sys_utilities.h: No such file or directory
 #include "sys_utilities.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

How can I fix it?

skieast, 2022-03-15:
I've made changes on my fork to build using Cmake and the latest stable idf (4.4) 
If there is interest i could do a PR

danjulio, 2022-03-15:
Yeah, that sounds interesting.  Would be good to get off of 4.0.2.  Things I'm worried about are having to retest everything and performance.  Can you give me a quick idea of what had to change?  (I saw your other issue regarding cJSON.  Not sure what happened there but it looks like cJSON should be part of every IDF release).

skieast, 2022-03-16:
From my memory:
I changed all the CmakeList.txt files to include all the dependent files. Seems to be needed to cmake correctly
Changed one line to switch the heap_caps_malloc region to MALLOC_CAP_SPIRAM for the json image text buffer. The IDF spi code checks if the buffer passed is in DMA region, if not the code allocates a buffer in DMA region and memcopies the result to the passed buffer.  I only compiled and tested briefly with the IOS app. 

	sys_image_rsp_buffer.bufferP = heap_caps_malloc(JSON_MAX_IMAGE_TEXT_LEN, MALLOC_CAP_SPIRAM);

I was originally only trying to build it so for a PR would just start from scratch and apply enough changes with nothing extraneous. 

Yes you are right about cJson.   As I am bulding with cmake the components being used in the IDF have to be explicitly listed. I'll make that change and that should be it.

danjulio, 2022-03-16:
Thanks for the info.  Why did you have to change the malloc for the sys_image_rsp_buffer?  Did the system run out of memory (malloc fail)?

As you figured out, that buffer needs to be in DMA accessible memory because that buffer is passed to the SPI Slave peripheral when the camera is used in the new Hardware Serial mode that the FW 2.X supports (allows connecting the camera directly to another micro instead of using Wifi and used by the tCam project).  It seems to me that the same out-of-memory issue might occur if the SPI driver has to allocate the same size buffer.  The copy may also impact performance since it will occur in series with data being available and notification to the external micro that an image is available.

I say make the PR when you can and I'll test it here.  I just can't promise it'll be done immediately.  But I do appreciate you making the effort and digging into, and understanding, someone else's code!

skieast, 2022-03-16:
Yes, the malloc failed. Doesn't appear to be enough space in dma accessible memory.  I didn't realize that the buffer in question is used for sending to a slave SPI device.  As you said that buffer must be in DMA memory.  The available largest free block of DMA memory to malloc appears to be about 33792 bytes (from what i remember as I was testing).  I believe that the code is trying to malloc 1204*53 or 54272 bytes.  So short a bit. 

I'll just get it to build with the idf.py (cmake) build system but it won't work with a slave peripheral and dma mode,  

As you said interesting looking at someone else's code and trying to understand it and what hardware is involved. Thanks for the responses.

skieast, 2022-03-17:
Just did a brief test with the attempt to allocate the DMA accessible memory for the spi buffer. Added a log for the free memory
```
I (1753) sys: Buffer Allocation
E (1753) sys: malloc shared json image text response buffer failed
E (1753) sys: DMA largest free block: 33792
E (1763) main: tCam Mini memory allocate failed
```

danjulio, 2022-03-17:
As you thought.  Something in 4.4 is taking more memory.  Off the top of my head the ways to deal with this are a) try to claw back enough internal memory with code changes so the buffer will fit again and b) see if there's anything that can be adjusted in the IDF through the config file. 

I forgot to mention that another thing moving the buffer to internal memory did was to improve performance.  The IOS App doesn't display FPS but the desktop application does.  Do you see a drop in performance with the buffer in PSRAM?

skieast, 2022-03-17:
OK got the memory needed. Moved one of the json buffers to spiram/psram. Seems that the memory allocator puts MALLOC_CAP_8BIT into internal ram instead of using the external ram first. Even though it is 8bit addressable. 
What sort of frame rate are you seeing with the desktop app over wifi? No idea if this change has affected anything. 

I'll now remove all my debug prints and try to get a PR together.

danjulio, 2022-03-18:
Well this is a bit embarrassing.  I took a peek at json_image_text (that I am guessing you moved up to PSRAM) and realized it was no longer being used (a hold-over from the 1.x code I didn't clean up).  So I think there isn't an internal memory space issue after all.  Sorry about that...(and thanks).

The 2.X FW should be able to get the full ~8.7  FPS the lepton is capable of.  Wifi variability can affect this a bit but you should be able to see the desktop app reporting around that FPS while streaming at the fastest rate.

skieast, 2022-03-18:
All good. Thanks again.  I'm seeing around 6fps but as you said wifi variability can affect this. I'll build a stock image without my various changes and compare.  And looking at code like yours is how i learn stuff. or attempt to.

closed 2022-10-03 by danjulio