danjulio/lepton #5: Hex-File for TTGO T7 1.5

kami83, 2021-04-23:
Hi, 

i am trying to use the firmware with the Lilygo TTGO T7 1.5 board. Can you offer me a precompiled hex-file and the correct wiring for this board?

I have got a lot of problem because i am using the Lepton 2.5 with the Breakoutboard 1.4.

Thanks a lot.

cu kami

danjulio, 2021-04-23:
Hello,

Precompiled Version 1 code in github repository should work for T7 1.5.

However, my code only supports Lepton 3.5.  I plan no support for any other module.  I don't have any other Leptons and I am not interested in them personally.   Sorry.

Regards, Dan

kami83, 2021-04-26:
Hi,

thanks a lot. I have also a 3.5 Lepton. So i can also test it with this module. Which wiring do you use. For example the SDA is on pin 23 in you picture, but on the module it is on pin 21??
Do i really need the VSYNC pin? on my breakout board there is no VSYNC pin.

Also another question can i get a tCam pcb board? 

Thanks a lot.

BG kami

danjulio, 2021-04-26:
The ESP32 is flexible about what functions are associated with various GPIO.  You need to following the wiring as I showed it and with the GPIO described in the code's system_config.h file (copied below).

You need VSYNC.  At least one of the earlier Group Gets breakout boards provided VSYNC as a pad on the PCB (GPIO3).  You can see this if you look at my code associated with the Teensy 3.2 in this repository and documented at [hackaday.io](https://hackaday.io/project/159615-lepton-35-thermal-imaging-camera).

Group Gets will carry the tCam-Mini assembled PCB.  

```
//
// IO Pins
//   Lepton uses HSPI (no MOSI)
//
#define BTN_IO            0
#define RED_LED_IO        14
#define GREEN_LED_IO      15

#define LEP_SCK_IO        5
#define LEP_CSN_IO        12
#define LEP_VSYNC_IO      13
#define LEP_MISO_IO       19
#define LEP_RESET_IO      21

#define I2C_MASTER_SDA_IO 23
#define I2C_MASTER_SCL_IO 22

```

kami83, 2021-04-29:
Hi,

thanks a lot for the answer. I found the Pad and will test it. But on the github history i found to folders with precompiled firmware in version 1.0. First the root folder precompiled with 3 Hex files and the subdirectory wih ESP32_v1_silicon. Which are the right version for my TTGO 1.5?

BG kami

danjulio, 2021-04-29:
I don't know what revision of the ESP32 chip is in your module.  You can start with the v1 firmware.  It should work.  And you can try the v3 firmware - the camera won't boot if the silicon is version 1.  You can look at the serial output (115200 baud) and the boot process will tell you the silicon revision.

kami83, 2021-04-30:
Hi,

thanks a lot. It works really fine with the TTGO 1.5 and the Lepton 3.5 (It doesnt work with the 2.5 Module). I have to use the Silicon v1 Version. If i use the AP setting i can connect to the standard ip with the Desktop App. But if i change the wifi setting to my router i can not change the IP Adress in the Desktop app, because i cannot write any . (dots) in the ip only ",". Maybe you can fix this??

Thanks a lot.

BG kami

danjulio, 2021-04-30:
I'm glad you got it running.  I do not understand your comment about entering an IP address.  I have tested that on all supported platforms and it works.  Try selecting the existing contents, deleting them and then entering an IPV4 address (e.g. XX.XX.XX.XX).  If you continue to have trouble then please provide more information so I can investigate.

kami83, 2021-05-01:
Hi, thanks a lot for the quick answer. I found a solution, but first of all i will try to describe the problem. In your mask Desktopapp under Win64 i can enter a ip adress in the field, if i use my keyboard i can not enter all numbers and i can not enter a dot. so i can only enter something like 100,100,100,100. The software is blocking this. I found a workaround. If i delete the field contents and then paste the IP adress it works. It looks like the field is filtering some values. Maybe you can fix this problem in your next version. 

I have got another question. I am would like to add a TFT and SD-slot for my TTGO. Have you got a schematic for this (like your tCam) and maybe the firmware? 

Thanks a lot. Have a nice weekend.

BG kami

danjulio, 2021-05-03:
I apologize but I cannot duplicate your failure with the IP entry on Windows.  I can successfully enter an IP address in many ways on both Win 7 and Win 10.  The field does attempt to keep the form "XXX.XXX.XXX.XXX" but you can enter any IP address if you select the existing contents to replace. You can also select any sub-field and replace that with no problem.

I have no plans to extend tCam-Mini with an LCD and SD-Card.  You might look at the tCam-Mini source code as well as other examples of using whatever LCD you want to use and then see how I generate visual data in my FireCam source code if you wanted to tackle this project on your own.

kami83, 2021-05-04:
Hi, i tried to make a screenshot. In field 1 i can only enter numbers with comma in this style. In all other fields 2,3 i can enter everything. Maybe you use a mask or a filter for the IP field?
Under Windows i can paste a normal ip adress. Under MacOS this is not possible.
Can you send me your XOJO Project? I would also like to see how you calculate temperatures etc. for some function in my python scripts.

For the functions with the LCD and SD-Card i will try to wire it with my TTGO. Have you compiled a firmware of FireCam for ESP32 TTGO?

Thanks a lot. Have a nice day.

BG kami
[image]

danjulio, 2021-05-07:
Ah, I see.  I didn't consider international settings.  Thanks for explaining.  You are correct that I used a mask.  I have to think about it but maybe I'll just remove it and try to validate later.   I'll let you know when there's a new version to try.

At this time I'm not sharing the source code for the xojo applications.  However I did upload demonstration python programs that show how to calculate spot temperatures from both telemetry and the CCI interface.

Unfortunately no Firecam firmware for anything other than the custom board I designed.

kami83, 2021-05-07:
Hi, thanks a lot for the answer. I will wait for the new version.

BG kami

closed 2021-05-07 by kami83