danjulio/lepton #6: Example for python with temperature and stream

kami83, 2021-05-03:
Hi,

can you please give a python example how to decode the temperatures from the radiometric string? 

Maybe you can also give an example how to show the stream with python?

Thanks a lot.

bg kami

danjulio, 2021-05-03:
The python library is a work in progress and is not my area of expertise.   The engineer who is helping me and I have plans for more demos.  I will certainly take your suggestion into account but I cannot promise to make any particular type of demo.

kami83, 2021-05-04:
Hi, no problem i am working on it. Temperature is done now i am trying to watch the stream. If i can i would like to help.

BG kami

danjulio, 2021-05-07:
I updated the python area with a new driver and set of examples.  You'll have to upgrade your camera's firmware to v1.3 as it has new features the driver is using.  One example displays an image and the others show how to decode the telemetry stream and communicate directly with the Lepton's CCI interface (be careful, easy to crash it).

kami83, 2022-05-10:
Hi, 

is it maybe now possible to give an example for video streaming with python?

BG kami

bitreaper, 2022-05-19:
Hi BG Kami, you should now be able to find a stream example in examples/streamtest.py.  It is using the Tk library which is the easiest to install and most widely available.  With pull request #26 you should be able to install tkinter and use it.