danjulio/lepton #7: How should I convert the stored radiometric files to a standard image output (jpg, png, gif)

baldeagle, 2021-08-02:


danjulio, 2021-08-02:
The desktop application can export these formats from an image directly acquired by the camera, from a file or from a frame in a video.  You use the preferences to select if you will add additional metadata to the image or not.  You can also include the spotmeter or not.

baldeagle, 2021-08-02:
I would like to train an object detector based on the stored radiometric images (.tjsn suffix).  But I'm not sure how to unpack that format into something that can be used by normal image detector algorithms.   I'm currently trying to modify dump_image.py to read from the stored files.

Also: Thank you for developing this and sharing it via groupgets!  It is an awesome little thermal monitor.

danjulio, 2021-08-02:
I should probably add a python example showing how to read a file, but possibly you could read the file, convert the base64 json image data into an array and then convert that into an RGB array which you could either directly use with your algorithm or convert it into a known filetype that they take.  I don't have examples showing how to read the file and but the examples show how to decode the base64 and get an RGB image.  I'm not very familiar with matplotlib but it can probably generate a known image file from the grayscale image that is calculated in disp_image.py.