danjulio/lepton #8: Fledgling R package

hrbrmstr, 2021-08-20:
Just dropping a note to, first, say many thanks for the tCam and this repo!

I've got a fledgling R package going for the tCam-Mini over here: https://github.com/hrbrmstr/tcam and will be adding more to it and likely writing a Go, Rust, and/or Swift command-line tool for the device as well.

danjulio, 2021-08-21:
Hi Bob, that is really cool.  In fact you're one of the kind of people I hoped to reach with tCam-mini.  People who would be interested in accessing the data with other tools for various kinds of analysis as opposed to just looking at images.  If you don't mind I will link to your repository on the main page.

hrbrmstr, 2021-08-23:
That'd be 👍🏽!