danjulio/lepton #10: interested in tcam

spereg53, 2021-09-27:
how can  integrate publish mqtt topics on esp32, for send the value  to some broker, ,the maxim distance of the lepton??? and the maximum grades on centigrades,that can measurement,i wan to make some proyects for electrical devices,in a substation electrical,thanks in advanced

danjulio, 2021-09-27:
Hello spereg53:

I'm not sure I understand your questions but I have attempted some answers below.

"how can integrate publish mqtt topics on esp32, for send the value to some broker"
Could you use the python driver and write some python code running on a computer that connects to the tCam-Mini and publishes the information you are interested in using a MQTT library?

"the maxim distance of the lepton???"
I do not understand this question

"the maximum grades on centigrades"
Temperature information can be had two ways:
   1. Use the spot meter data in the telemetry
   2. Compute a temperature anywhere by taking the radiometric data and converting it to temperature using the gain information in the telemetry.
   
  You can see examples of how to use the python driver in the repository as well.

spereg53, 2021-09-27:
do you know some flow for working on node-red???thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

danjulio, 2021-09-27:
No, I'm sorry.  I don't have experience with node-red.

Thovthe, 2022-03-18:
I think this issue can probably be closed but I'll add this to cover at least one of these questions.

To know about the technical capabilities on the sensor (max temperature in different modes etc.) look at the data sheets linked from these pages:
https://store.groupgets.com/products/flir-lepton-3-5
https://store.groupgets.com/products/flir-lepton-2-5

closed 2022-03-21 by danjulio