danjulio/lepton #12: ESP32 Python FFC Mode

jperfo, 2021-11-21:
Hello,

at first thank you for the wonderful project!

Because I want to integrate the tcam mini in an house alarm system with motion detection I encountered the automatic FFC to be a problem. A problem, because on the FFC action the motion detection is triggered. If I could get the FFC status oder control it manually this could be resolved. Therefore my question:

1. Is it possible to read the FFC status via Python?
2. Is it possible to set the FFC mode to manually via Python?
3. If all the above is not possible: What would you recommend? :-)

Thank you very much
jperfo

danjulio, 2021-11-21:
Thank you for the kind words.  I understand how the FFC might confuse a motion detection algorithm (and the few seconds after as the Lepton settles back down).  It should be possible to do what you want using the python interface.  You should download, if you have not already, a copy of the Lepton data sheet (which describes the contents of the telemetry data) and the Interface Description document (IDD - which describes the CCI command interface).  You should be able to find these on the Group Gets Lepton pages.  Lots of good information about the Lepton and the way it works.

1. There are two ways to read the FFC status.  The Status field in the telemetry data returned with each image contains current FFC information in words 3 & 4.  See "Table 3 - Status Bit Encoding (Telemetry Row A, words 3 and 4)" in the Lepton datasheet.  You can also use the python driver's get_lep_cci method to run a Get SYS FFC Status command (see 4.4.29) in the IDD.

2. You can configure the Lepton's FFC mode using the python driver's set_lep_cci method to run a Set SYS FFC Mode Control command (see 4.4.27) to configure the mode.   Then you can use the SYS Run FFC Normalization command to explicitly execute a FFC if you want.

The get/set_lep_cci methods provide direct access to the CCI and are configured in terms of the 16-bit word data fields so you have to map the IDD C-language description into the actual 16-bit data words sent/received from the Lepton.  The intent of these methods was to be able to execute commands that there wasn't a more specific python API for.   The Desktop application also provides a way to run direct CCI commands and might be useful to poke around and see what the lepton returns, etc.  Of course you can do this from python as well.  The "cci_demo.py" program shows how to use these methods.

I hope this is helpful.  Good luck with your project.

baldeagle, 2022-01-12:
How are you integrating the TCAM into you home monitoring setup?

danjulio, 2022-01-12:
I haven't.  I did write a toy web server that is designed to connect to one or more cameras and store images when certain temperature criteria are met.  I played with it last fall looking for large animals on my property with mixed success.  What I want to do is exactly what you're doing and add motion detection but it seems a non-trivial problem to solve.

jperfo, 2022-01-14:
Hello,

at first, thank you very much, danjulio. Sorry for the late answer but I was a bit busy.

Your advice was very helpful so I could solve the problem regarding FFC. The current version of the motion detection script works with OpenCV. When the FFC is triggered the script ignores the changing pixel for a certain amount of time. An alternative would be to trigger FFC manually or to deactivate. However, I try to avoid the latter. 
What of course needs to be said is the fact that the are of surveillance is very limited in my use case - it's a balkony. I haven't integrated it fully yet but the test result were ok. One must play with various settings that can be done for such a script, the alarm should not be triggered as soon as there is a human detected (because it could be also something different) but after a certain amount of time. Another approach worth trying out could be a tensor flow trained alarm, eventually.

I can keep you informed as soon as I make progress.