danjulio/lepton #13: Feature Request: Android App

gigawatts, 2021-12-15:
Any plans in the works for an Android app to stream video and take snapshots from the tCam-mini? That would make for an excellent mobile setup, with the tCam-mini in AP mode, phone/tablet connected to the AP and no PC required.

danjulio, 2021-12-15:
Short answer: yes!  I want it, too.

Long Answer: I have to learn to write mobile apps and my development platform, xojo, for which I have a ton if libraries and supporting code for tCam has to support Sockets for the Android build system (they just released their alpha Android build system).  I'm going to start with IOS since it's fully supported but hopefully the code I write for that will port right over to Android when xojo is ready.  I wish I could give you a date but there are too many variables.  Just be assured I'm interested in this too.

gigawatts, 2021-12-16:
Awesome! If you need any beta testers for Android, let me know. I don't own any iOS devices made in the last decade, so I probably won't be of much use there, haha!

KingRikkie, 2022-02-09:
.NET MAUI (former called Xamarin - full-feature cross-platform mobile & desktop framework) comes to mind. One code base + shared UI can generate builds for Windows/Mac/Linux/iOS/Android all at once.

danjulio, 2022-02-21:
The iOS App just went live at https://apps.apple.com/us/app/tcamview/id1610451813

As soon as xojo supports sockets for Android I will begin porting it there too!

@KingRikkie - I appreciate the suggestion.  Unfortunately I do this on the side and have a big investment in xojo code already (that has proven to be very useful).  I don't have the time to take on a new development environment.

dhelleberg, 2022-10-17:
Is the source code of the Xojo app available somewhere? I could take a look at it and may support android porting....

danjulio, 2023-02-24:
Another developer has created an Android app and is looking for alpha testers.  He's using the Play Store testing mechanism to distribute the app so please email me with your Google Email ID (for the phone) if you'd like to test the app.