danjulio/lepton #14: Quick explanation of I2C port on tCam-Mini

timbo100, 2021-12-21:
Hi Dan,   I'm very thankful for your project and I bought a batch of the tCam-Mini PCB's from GroupGets.  Thanks

I'm looking forward to digging into the hardware design and software components but I'm a little too busy and need a quick(er) answer if you don't mind.

I would like to integrate/communicate with a Lepton 3.5 from another SBC which may not have an OTG USB port and so trying to figure out how I will be able to get access radiometry info.  I see an I2C interface and was wondering what's accessible from that I/F?

I'm hoping to be able to query temperatures across the 2D pixels array from another ESP32, if not the onboard ESP32.  Any thoughts/suggestions?  

Again, great job and thanks

danjulio, 2021-12-21:
Hello,

The I2C port provides access to the bus used by the ESP32 as master communicating with the Lepton's Command and Control Interface (CCI) and is how it configures the Lepton.  I broke it out because I had been thinking of including support for a I2C-based TOF distance sensor to try to improve accuracy but some testing shows that there may not be a big gain doing that so that part of this project moved to a lower priority for me.  Unfortunately it cannot be used by an external device to communicate with tCam-Mini.

Primarily tCam-Mini is designed to be accessed via the Wifi interface.  The new version 2.0 Firmware has support for a new hardware interface on some of the unused ESP32 IO pins that can be used instead of the WiFi interface.  I will use this for the tCam project but it could be also be used with another controller.  It consists of a serial port over which the same json commands are sent and received as over the Wifi interface with the exception of the image json object which is retrieved over a separate SPI slave interface (since there is so much data).  I still have to document this interface but it is functional in the new firmware.

A future revision of the tCam-Mini PCB will break these signals out to a header, although with delicate soldering it is possible to connect the necessary wires directly to the ESP32 module pins on existing boards (this is what I did for testing).

I will try to spend some time in the next couple/three weeks to documenting this interface in the tCam-Mini github readme.

FYI: tCam-Mini does not support image data transfer on the USB interface (like Group Get's PureThermal board which looks like a UVC class camera).  This interface is used for power and programming the ESP32 as well as monitoring diagnostic information over USB Serial.  You will need to use a Windows based utility (or the Espressif SDK) to load the 2.0 firmware but after that the new firmware also supports OTA updates from the desktop application for subsequent updates.

timbo100, 2021-12-21:
I appreciate your quick feedback.  I will look at the possibility of a local wireless connection, or there's also the possibility that I rewrite your ESP32 code to do what I need.   Thanks again.

closed 2021-12-22 by danjulio