danjulio/lepton #17: [FR] tCam + tCam-mini merge

Ramalama2, 2022-02-11:
Hey dan, im pretty sure this was asked before somewhere, but anyway.

First, the tCam mini is a really nice project, and probably the only way to make the lepton 3.5 usable. So i don't really understand why there aren't any other (halfway actual) projects about this camera.
However at this point, thank you very much for all the effort you do!

Now the main point, that i don't 100% understand...
Somehow tCam and tCam mini are 2 separate projects, with different hardware etc.... (At least from what i see)

Wouldn't it make more sense to use always tCam mini as the "carrier board" for the lepton 3.5 and add to it an wired interface that will allow it to connect simply to an raspberry?
I know the usb serial interface is probably not fast enough, so you need to find only a solution for that.
But this would allow it, to make everything much simpler, as the "full" tCam will become only a set of:
Rpi4 + desktop app
tCam-mini
Some sort of battery management pcb

And if you have a full tCam "set", it will still allow for a wifi access, because of tCam-mini. (Or if there isn't space on the esp, the raspberry could make an hotspot/nginx server either...)

And especially as the most interesting part of this is, this would allow every user to start with a tCam-mini and upgrade it later with almost no effort to a full tCam 🙈

Cheers ✌️

danjulio, 2022-02-11:
Hey, the answer to your questions of why is history.  Simply how things unfolded as I played around with the Lepton.  :-)

I started with the full ESP32 tCam but it ran into performance and complexity issues so I spun out tCam-Mini with the Lepton and Wifi parts as a first product.  I then re-structured the original tCam code into a part that handles a display (w/ GUI) and SD-card storage plus more commands.  To support this tCam-Mini got some new functionality that is actually along the lines of what you propose (see below).  I also scratched an itch to design the ESP32 dev board (gCore 2) that I wanted with a very high performance 480x320 pixel cap-touch display, good power and battery management, RTC and fast SD-card storage.  So now tCam = tCam-Mini + tCam firmware running on my (yet to be introduced) gCore dev board and it's pretty much done.  I just have to document everything and find a way to get some gCore boards built (at this writing, parts shortage is a killer).

Before I played with the ESP32, I experimented with the Pi (getting data reliably from Lepton was pretty iffy at the time - I always thought it needed a proper kernel level driver) and then using the PRUs on a Beaglebone along with a hacked version of another power board I had made for the Pi (Solar Pi Platter).  But I more-or-less lost interest in that project because of how slow Linux booted on the Beaglebone black and the work I'd have to do to the PRU code to get full radiometric data out of the Lepton (on the other hand the PRUs are a rock solid way to do real-time stuff like reading the Lepton).

Starting with FW 2.0, tCam-Mini now sports a new hardware interface (you can see it if you look at the code).  The interface is comprised of a serial port running at 230,400 baud and a slave SPI port.  A Mode input tells the camera on boot to use either the Wifi interface or the hardware interface (serial mode).  When configured for serial mode all the same json commands that are sent and received over the wifi interface can be sent and received over the serial interface with one exception.  Instead of sending image data over the serial interface the camera sends a new "image_ready" message over the serial interface and the external controller can read the image using the much faster SPI port.  This is how the gCore communicates with tCam-Mini to make tCam and it's certainly how another controller like a Pi could use tCam-Mini.  The only missing thing is a library to make using it easier.  Perhaps the python driver could be modified to use serial/spi peripherals instead of a socket.

I have designed a new version of the PCB that breaks these signals out to a header and it should be available within a few weeks on Group Gets.  Intrepid users of the current PCB revision could actually solder wires to the castellated connectors on the ESP32 like I did during development.

In the next few weeks I will have all of this documented and the new hardware revision released.

So if you made it all the way here I hope this helps explain why things ended up the way they did and what I'm going to do going forward.

Ramalama2, 2022-02-11:
Now, while you mentioned the serial/spi connection, i remembered again, that i readed it already somewhere here on GitHub 🙈

However, thank you very much for all the explanation and im glad that you basically confirmed, that we can use the tCam mini later together with some sort of an sbc that has spi, like the pi.
This will basically allow to design/make (3d print) an portable tCam. At some point in the future at least, once the software hurdles are taken.

However, sadly im not a dev, but i can help out with cad/modeling if there is need.
Thanks again dan 👍✌️

PS: should i close this issue?

danjulio, 2022-02-14:
I'll close this.  I hope the new board is helpful to your project when it's released!  Good luck with everything.

closed 2022-02-14 by danjulio