danjulio/lepton #20: Packaging and Licensing

Thovthe, 2022-03-17:
Very cool project, I have a t-cam mini on the way.

I thought I'd make up a pkgbuild of the applications for use on arch based distributions but I can't seem to find the source files or any build instructions.  Are those components not intended to be open source?  I'm also not seeing licensing information for many of the other files.  I'm not sure that I can do much more than packaging and letting people know about the project but I wanted to contribute.

danjulio, 2022-03-18:
Thank you.  And thanks for offering to make a pkgbuild and contribute.

Currently the supporting desktop and mobile applications aren't open source.  Partially because I use the paid xojo development environment and most people have no experience with that or would want to pay for it.  Partially because I worry about the support load of people digging into that code, and partially because I wanted to reserve the ability to possibly make modified versions of it for pay.

I do support the idea of open source - and have published the source to a ton of my work - although I am not religious about it.  Many parts of this project are open source (such as the firmware).  I am willing to have some discussion about sharing the source at some point.

Regarding the licensing: Yeah, probably not super clear and I'll put it on the list of things to do when there's time.  At least a note in some of the readmes.

Maybe not the answer you wanted to hear, but it is the honest truth.  I hope you still find value in the work I've done.

Thovthe, 2022-03-18:
Those are 3 reasonable concerns and I respect that it is your work that you may treat as you see fit.  I hope you won't mind me following up on some points though.
<br />
Is all the business logic for the webapp and the desktop app in xojo's language as well?
<br />
As long as you keep track (maybe with a separate repo) of what code is originally yours and what are community contributions you'd still be free to do whatever and license however with what is your work.
<br />
Having it be FOSS doesn't mean you couldn't get whoever is asking for these other features to pay you to develop them.  Certain licenses would even allow you to use community contributions in closed source as well as open source copies such as 3-clause BSD (though because I am somewhat religious about these things I look at them funny :wink: ).
<br />
I'll leave this issue open till there is more clear licensing for the open source components.  **Sound good?**
(If you clearly outline it some other way I might make up a pull request with the right words in the right places.)

danjulio, 2022-03-21:
All code in the desktop, web and mobile apps is created using xojo.

I appreciate your input regarding licensing.  It is something to consider.

In your opinion, what would make the licensing of open source components more clear?  For example the tCam-Mini firmware clearly states the license in every source file.  Should there be a comment in the readme for that piece?