danjulio/lepton #22: tCam-Mini with Serial Output

johannesosterberg, 2022-04-06:
Hi, I am excited about my tCam-Mini, I am in the process of building a small Pan Tilt Mount.

Is it possible to add something to the firmware to control the serial interface? 

This way a small Arduino could control the Pan Tilt Mount using serial commands and I don't need an extra cable or wifi connection. 

Four buttons could send serial commands for up, down, left and right to the Arduino would also be great. Is something like this planned or even feasible? That would make the tCam-Mini even more exciting. 

The Arduino code be very simple to implement.

Thank you and many greetings

danjulio, 2022-04-06:
Sounds fun!  Unfortunately, based on your description, I'm not entirely sure what you are asking.  For example where are the buttons and what serial interface?  Is an arduino controlling to the servos/motors?  If so, how do you envision it communicating with tCam-Mini?

The primary serial interface is connected to the USB UART so probably can't be used for anything else (theoretically the ESP32 TX output could talk to something but those signals are hidden in the PCB and not available).

Starting with FW 2.0 there is a (currently undocumented) new hardware interface that makes use of the second serial port in the ESP32 and will be available on future PCBs but this is used for a new mode that allows an external controller to communicate with tCam-Mini directly instead of using WiFi.

Finally there is the internal I2C bus which is exposed on an expansion interface.  This is used by the firmware to configure the Lepton through its CCI interface.   I would use this interface if I were adding an expansion function to tCam-Mini.  It would communicate with some external controller like an Arduino that presents itself as an I2C slave. Then you could add support for an additional json command packet to the FW which could be used to deliver custom data via the I2C interface (using the FW's internal I2C API).  You'd also need custom code on the other end capable of sending this packet (e.g. none of the apps I've written to support tCam-Mini would know anything about this new command).

It doesn't make sense for me to add one-off functionality for individual users of the camera but I do encourage you to consider making your own custom firmware.  As I mentioned above you'd also have to make a custom program to control the camera and view images from it (but this could be facilitated using a modified copy of the json library with your new command added).

johannesosterberg, 2022-04-07:
Thanks for the quick answer, I hadn't realised that the USB UART overrides the serial interface. Thanks for the many points of reference, I'll have to think about how best to do this. Your code is not public, is it? I can understand that you don't want to incorporate the feature. Thanks anyway, I'll have to think about how to solve the problem for myself.

closed 2022-04-14 by danjulio