danjulio/lepton #23: transfer stop while ffc

tantalar, 2022-04-12:
Hello,
 This is very interesting project and would like to work with this. I am using pocketbeagle plus qqvga LCD (ST7735) and the code seems working pretty well. But you know, sometimes image transfer stops (Lepton freezes) and would like to solve this problem.

Before that, I see transfer stops (not freeze, CSN is de-asserted and PRU0/1 LEDs are off) when Lepton executes FFC automatically. I don't see any code handling FFC, it is under auto mode. It is separated from the image path (capture using PRU) then would like to know why it happens (or commanded to PRU from OS? how?). I think image capture must not stop while executing FFC...

danjulio, 2022-04-13:
Hey, glad to see this old code being played with!

In this system the Lepton is deciding to perform the FFC automatically (when the its temperature changes or every few minutes).  Unfortunately the Lepton is unable to generate data while it is performing a FFC.  What is happening is that the code running on PRU0 cannot get a valid frame during the FFC (it is incrementing lep_resync_count) so it de-asserted CSN for LEP_RESYNC_CYCLES (~190 mSec) to reset the Lepton VoSPI.

tantalar, 2022-04-14:
So, when FFC, what is the return value of get_packet()? Is it PKT_DISCARD?

danjulio, 2022-04-15:
Yes, probably PKT_DISCARD or perhaps PKT_ILLEGAL.

tantalar, 2022-04-15:
Thank you for your help.
I would like to ran this interesting code 24H continuously....will update when I succeed.

closed 2022-04-15 by tantalar