danjulio/lepton #32: Connect external antenna on tCam-Mini

GillesBra, 2022-07-19:
Dear Dan,
I have bought the tCam-Mini Rev 2 board along with a Lepton 3.5 camera. My model is with a built-in antenna.
I would like to connect an external antenna to increase the range of the wifi connection but I am not sure this is possible with my version of the ESP32 (in the BOM there are 2 different references for built-in or external antenna).
Is it possible to connect an external antenna with my model ?
Which connector reference must I choose and how do I connect it to the board ?
Kind regards.
Gilles

danjulio, 2022-07-19:
Hello Gilles,

It is technically possible to add an external antenna to the WROVER module on tCam-Mini - but it's delicate and you risk destroying the antenna connection and rendering the device useless.

Basically you have to move the 0-ohm resistor that connects the antenna output to either the built-in PCB antenna or an IPEX connector pad.  Then you solder an IPEX connector to the board and connect to an external 2.4 GHz antenna.

The gang at Random Nerd Tutorials has a blog post showing how to do this with the WROOM module, but the process is the same with the WROVER module used on tCam-Mini.

https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/

I was able to do this on a prototype but it is pretty delicate work and you should be prepared to lose your tCam-Mini in the process.  The risk is that you pull a pad off the board while removing the resistor.  I ended up losing the tiny resistor and had to replace it with a tiny piece of wire.  I have no idea how this affected signal quality.

So consider yourself warned - but good luck if you choose to try this modification.  There's always a new Rev 4 board waiting for you to buy at Group Gets should you fail.  :-)

Regards, Dan

GillesBra, 2022-07-20:
Thank you very much Dan.
The process is very clear, but, you are right, it seems risky because the
connections are very small.
I think I'd better buy a Rev4 board.
Kind regards.
Gilles