danjulio/lepton #33: Building with idf.py fails

project-x51, 2022-07-25:
Hi Mate,

I love the work your doing with the Lepton.

I cloned your repository and tried to build it inside the VSCode terminal (Espressif Plugin for VSCode) but it failed due to some file dependencies. I was able to fix some of them by adding the REQUIRES keyword into the CMakeLists.txt file but some files appear to be missing from the repository all together.

../components/cmd/upd_utilities.c:27:10: fatal error: esp_ota_ops.h: No such file or directory
 #include "**esp_ota_ops.h**"

../components/sys/ps_utilities.c:30:10: fatal error: nvs_flash.h: No such file or directory
 #include "**nvs_flash.h**"

../components/cmd/json_utilities.h:41:10: fatal error: cJSON.h: No such file or directory
 #include "**cJSON.h**"

These are the changes I made to the CMakeLists.txt files to resolve the other issues.

**components/cmd/CMakeLists.txt**
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES clock lepton sys)

**components/lepton/CMakeLists.txt**
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES i2c sys)

**components/sys/CMakeLists.txt**
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES cmd)

Cheers!

project-x51, 2022-07-25:
I'm using the ESP-IDF 4.4.1 BTW

skieast, 2022-07-26:
I went through the same process and did eventually get it working,  I was working on a PR but put it aside due to other pressures. I  will have to look at it again but you are on the right track.

danjulio, 2022-07-27:
Hey project-x51.  Thanks for the fixes to my CMakeLists.txt files.  I will incorporate those.

I'm not sure why you're not seeing the IDF header files.  I checked and they exist in 4.4 as well.  I still build using the old make system and perhaps you have to be more specific about specifying the IDF directories with the new system.  Maybe something else in the local CMakeLists.txt file or you have to add something so VSCode knows to look in the IDF directories.

It's on my list to try to move the code base to a newer IDF version but the environment I have works so it's really hard for me to prioritize it over all the other things I want to do related to this project.

project-x51, 2022-08-08:
Hi danjulio,

OK, I finally got the code to compile. The CMakeLists.txt files needed to be edited to refer to all the required components and point to the ESP-IDF component folders. The one in the main folder also needed to point to all the C files there to ensure the linker could find them as well.

**CMakeLists.txt**
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/components")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(tCamMini)

**main/CMakeLists.txt**
set(SOURCES main.c ctrl_task.c lep_task.c mon_task.c rsp_task.c sif_cmd_task.c wifi_cmd_task.c)
idf_component_register(SRCS ${SOURCES}
                    INCLUDE_DIRS .
                    REQUIRES clock cmd i2c lepton sys)

target_compile_definitions(${COMPONENT_LIB} PRIVATE LV_CONF_INCLUDE_SIMPLE=1)

**sys/CMakeLists.txt**
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES cmd nvs_flash)

**cmd/CMakeLists.txt**
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES clock lepton sys json app_update)

project-x51, 2022-08-08:
Hi danjulio,

I was able to flash and run the code but then I get a couple of errors. I see if I can figure them out but there here for your reference.

PS D:\ESP32\Lepton\ESP32\tCam-Mini\firmware> C:\esp\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\esp\esp-idf\tools\idf.py -p COM14 monitor
Executing action: monitor
Running idf_monitor in directory d:\esp32\lepton\esp32\tcam-mini\firmware
Executing "C:\esp\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\esp\esp-idf\tools/idf_monitor.py -p COM14 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 d:\esp32\lepton\esp32\tcam-mini\firmware\build\tCamMini.elf -m 'C:\esp\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\esp\esp-idf\tools\idf.py' '-p' 'COM14'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM14 instead...
--- idf_monitor on \\.\COM14 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (29) boot: compile time 20:06:34
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ae28h (110120) map
I (151) esp_image: segment 1: paddr=0002ae50 vaddr=3ffb0000 size=039dch ( 14812) load
I (157) esp_image: segment 2: paddr=0002e834 vaddr=40080000 size=017e4h (  6116) load
I (160) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=864d8h (550104) map
I (364) esp_image: segment 4: paddr=000b6500 vaddr=400817e4 size=16314h ( 90900) load
I (402) esp_image: segment 5: paddr=000cc81c vaddr=50000000 size=00010h (    16) load
I (414) boot: Loaded app from partition at offset 0x10000
I (414) boot: Disabling RNG early entropy source...
I (425) cpu_start: Pro cpu up.
I (426) cpu_start: Starting app cpu, entry point is 0x40081420
0x40081420: call_start_cpu1 at C:/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (442) cpu_start: Pro cpu start user code
I (442) cpu_start: cpu freq: 160000000
I (442) cpu_start: Application information:
I (446) cpu_start: Project name:     tCamMini
I (451) cpu_start: App version:      2.1
I (456) cpu_start: Compile time:     Aug  8 2022 11:53:13
I (462) cpu_start: ELF file SHA256:  ea87008092093216...
I (468) cpu_start: ESP-IDF:          v4.4.1
I (473) heap_init: Initializing. RAM available for dynamic allocation:
I (480) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (486) heap_init: At 3FFC45A0 len 0001BA60 (110 KiB): DRAM
I (492) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (499) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (505) heap_init: At 40097AF8 len 00008508 (33 KiB): IRAM
I (513) spi_flash: detected chip: gd
I (516) spi_flash: flash io: dio
I (521) cpu_start: Starting scheduler on PRO CPU.
[... 23 lines cut ...]
I (710) wifi:Init dynamic tx buffer num: 32
I (720) wifi:Init static rx buffer size: 1600
I (720) wifi:Init static rx buffer num: 10
I (730) wifi:Init dynamic rx buffer num: 32
I (730) wifi_init: rx ba win: 6
I (730) wifi_init: tcpip mbox: 32
I (740) wifi_init: udp mbox: 6
I (740) wifi_init: tcp mbox: 6
I (750) wifi_init: tcp tx win: 5744
I (750) wifi_init: tcp rx win: 5744
I (750) wifi_init: tcp mss: 1440
I (760) wifi_init: WiFi IRAM OP enabled
I (760) wifi_init: WiFi RX IRAM OP enabled
I (770) wifi:set country: cc=US schan=1 nchan=11 policy=0

I (770) wifi_utilities: WiFi initialized
I (780) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (880) wifi:mode : softAP (7c:9e:bd:91:62:1d)
I (880) wifi:Total power save buffer number: 16
I (880) wifi:Init max length of beacon: 752/752
I (880) wifi:Init max length of beacon: 752/752
I (890) wifi_utilities: WiFi AP tCam-Mini-621D enabled
I (900) sys: Buffer Allocation
**E (900) sys: malloc RSP lepton shared image buffer 0 failed
E (900) main: tCam Mini memory allocate failed**

project-x51, 2022-08-08:
Hi Danjulio,

OK, I had some fun figuring this one out. I'm starting to get a bit more comfortable with the ESP32 now so it wasn't as difficult to solve. 

I grabbed the latest build from the master branch as I noticed it was updated just yesterday. This solved the memory allocation issue which I had figured out was because the module never looked for PSRAM. The I2C issue was a little more subtle and related to a change to the config for the I2C controller made in a recent revision of the ESP-IDF.

This is the revised code from the i2c component. It includes the clk_flags property which for some unknown reason is needed in ESP-IDF 4.4.1 which I'm using. I also added some error checking around the i2c_param_config call as this was unhandled previously.

```
esp_err_t i2c_master_init(int scl_pin, int sda_pin)
{
    int i2c_master_port = I2C_MASTER_NUM;
    i2c_config_t conf;

    // Create a mutex for thread safety
    i2c_mutex = xSemaphoreCreateMutex();

    // Configure the I2C controller in master mode using the pins provided
    conf.mode = I2C_MODE_MASTER;
    conf.sda_io_num = sda_pin;
    conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf.scl_io_num = scl_pin;
    conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf.master.clk_speed = I2C_MASTER_FREQ_HZ;
    conf.clk_flags = 0;
    esp_err_t err = i2c_param_config(i2c_master_port, &conf);
    if (err != ESP_OK)
        return err;

    // Install the I2C driver
    return i2c_driver_install(i2c_master_port, conf.mode,
                              I2C_MASTER_RX_BUF_LEN,
                              I2C_MASTER_TX_BUF_LEN, 0);
}
```

project-x51, 2022-08-08:
I also had to make some additional changes to the CMakeLists.txt files as the latest version references the mdns component and has replaced the WiFi_Cmd_task with the net_cmd_task. 

I'll create a pull request with all these changes for your reference.

**main/MakeLists.txt**
```
set(SOURCES main.c ctrl_task.c lep_task.c mon_task.c net_cmd_task.c rsp_task.c sif_cmd_task.c)
idf_component_register(SRCS ${SOURCES}
                    INCLUDE_DIRS .
                    REQUIRES clock cmd i2c lepton sys)

target_compile_definitions(${COMPONENT_LIB} PRIVATE LV_CONF_INCLUDE_SIMPLE=1)
```
**cmd/CMakeLists.txt**
```
file(GLOB SOURCES *.c)

idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS . ../../main
                       REQUIRES clock lepton sys json app_update mdns)

```

calvfoo, 2022-10-01:
hi,

manage to compile all (with new source codes downloaded today from github), but encountered this:

C:/esp-idf/components/tflite-lib/tensorflow/lite/experimental/microfrontend/lib/filterbank_util.c:99:28: warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]
   state->start_index = 1.5 + config->lower_band_limit / hz_per_sbin;
                            ^
[1235/1266] Building C object esp-idf/sys/CMakeFiles/__idf_sys.dir/eth_utilities.c.obj
In file included from C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c:35:
C:/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:15:2: warning: #warning "This header is deprecated, please use new network related API in esp_netif.h" [-Wcpp]
 #warning "This header is deprecated, please use new network related API in esp_netif.h"
  ^~~~~~~
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c: In function 'eth_init':
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c:88:2: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
  tcpip_adapter_init();
  ^~~~~~~~~~~~~~~~~~
In file included from C:/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from C:/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from C:/esp-idf/components/esp_event/include/esp_event.h:28,
                 from C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c:29:
C:/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c: In function 'ip_event_handler':
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/eth_utilities.c:261:43: warning: initialization of 'const tcpip_adapter_ip_info_t *' {aka 'const struct <anonymous> *'} from incompatible pointer type 'esp_netif_ip_info_t *' {aka 'struct <anonymous> *'} [-Wincompatible-pointer-types]
  const tcpip_adapter_ip_info_t *ip_info = &event->ip_info;
                                           ^
[1239/1266] Building C object esp-idf/sys/CMakeFiles/__idf_sys.dir/wifi_utilities.c.obj
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/wifi_utilities.c: In function 'wifi_init':
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/wifi_utilities.c:100:2: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
  tcpip_adapter_init();
  ^~~~~~~~~~~~~~~~~~
In file included from C:/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from C:/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from C:/esp-idf/components/esp_event/include/esp_event.h:28,
                 from C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/wifi_utilities.c:30:
C:/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/wifi_utilities.c: In function 'ip_event_handler':
C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/components/sys/wifi_utilities.c:467:43: warning: initialization of 'const tcpip_adapter_ip_info_t *' {aka 'const struct <anonymous> *'} from incompatible pointer type 'esp_netif_ip_info_t *' {aka 'struct <anonymous> *'} [-Wincompatible-pointer-types]
  const tcpip_adapter_ip_info_t *ip_info = &event->ip_info;
                                           ^
[1264/1266] Linking CXX executable tCamMini.elf
FAILED: tCamMini.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\tCamMini.elf.rsp -o tCamMini.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x14): undefined reference to `ctrl_task'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x30): undefined reference to `sif_cmd_task'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x3c): undefined reference to `rsp_task'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x48): undefined reference to `lep_task'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x50): undefined reference to `net_cmd_task'
[... cut ...]
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/main/main.c:67: undefined reference to `ctrl_set_fault_type'
c:/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Projects/lepton-master_202210/ESP32/tCam-Mini/firmware/main/main.c:70: undefined reference to `ctrl_set_fault_type'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

calvfoo, 2022-10-01:
anyone got the same error? or i did something wrong? i am using esp idf 4.4.1

calvfoo, 2022-10-01:
ok, it works now after i use this for cmakelist.txt in the main folder 

set(SOURCES main.c)
idf_component_register(
	SRCS 
		"main.c"
		"ctrl_task.c"
		"mon_task.c"
		"lep_task.c"
		"net_cmd_task.c"
		"rsp_task.c"
		"sif_cmd_task.c"

                    INCLUDE_DIRS "."
                    REQUIRES clock cmd i2c lepton sys)

target_compile_definitions(${COMPONENT_LIB} PRIVATE LV_CONF_INCL

danjulio, 2022-10-02:
Well done calvfoo.  I hadn't kept the cmake files up-to-date over the project. Thanks, in part, to project-x51's work, I just committed official version 3.1 of the firmware to the repository and it is now built using v4.4.2 of the IDF.  Aside from the build environment, the biggest changes were moving to the netif API (removing the deprecation warnings).  One thing for those of you who moved v3.0 to IDF v4.4.X is that performance and memory use changed some so make sure everything works as you expect.

closed 2022-10-02 by danjulio