danjulio/lepton #38: Questions about the wifi mode of tCam-Mini

GillesBra, 2022-11-28:
Hi Dan, I have purchased a tCam-mini Rev 4 with an external antenna in order to get the maximal range for a drone application in a student project. I use it with a Lepton 3.5. We have reached 120m but we wish to increase it as much as possible (500 m is the objective). So I have some questions about the design of the tCam-Mini:
1) What is the data rate in transmission with a Lepton 3.5 (theory gives me 2.8 Mbps )
a) in video mode
b) during image transmission
2) What is the wifi mode (802.11 b, g, n) used in these transmissions and what is the nominal RF power ?
Thank you for developping nice products.
Best regards
Gilles Brazzini

danjulio, 2022-11-30:
Hello Gilles,

Each image json packet is about 55 kBytes of data so for 8.7 FPS you're looking at about 3.8 Mbps at the data level (obviously raw data rate will be higher). 

Unfortunately much of the Espressif WiFi stack is a mystery to me.  I generally configure it with default settings.  According to the IDF website (https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-guides/wifi.html) the stack can support up to 20 MBit/s TCP throughput on 802.11b, g and n networks (I suppose it can't support 20 MBit/s on 802.11b).  According to the ESP32 hardware datasheet the typical TX output power for n (and g?) is 13 dBm and for b is 19.5 dBm.  I guess when it acts as a station it selects a mode based on the AP's mode but I'm not sure what the default mode is when it acts as an AP.

Based on my general experience extending WiFi range the antenna is one of the most important components and you could certainly experiment there.  I don't know if it's a possibility but perhaps using a slightly directional ground antenna (that you'd have to keep aimed at the drone) would help quite a bit.  In addition making sure you're running on as clean of a channel as possible is always helpful.  By default I am using channel 1.

From a code perspective there might be two places to look if you're willing to build your own binaries:

1. You can modify my default configuration in the components/sys/wifi_utilities.c file.  Look at the IDF documentation at https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html, in particular the configuration data structures.  You can force the WiFi type, data rate and max TX power. 

2. You might be able to use the configurator (menuconfig) to modify the WiFi stack configuration (for example to force various routines and data structures into IRAM).  However IRAM is slightly tight since I'm already trying to put performance sensitive buffers in there instead of the slower PSRAM so it's possible the code either won't compile or your could experience out-of-memory crashes while it's running.

Possibly an even more extreme solution would be to use Espressif's Long Range (LR) mode which claims up to 1 km range.  But that would probably take more significant changes to the code and (I think) you'd also have to use a ESP32 as the base station (also in LR mode).  This is described in the first link I sent you.

Good luck!  I certainly understand the desire for more range.

GillesBra, 2022-12-08:
Thank you very much Dan for this information.
Modifying the code is probably difficult and risky for us. But we will investigate this way.
Extending the antenna range is certainly a good solution for a 5/56 db increase.
I was also thinking of adding a RF amplifier at the output of the ESP32. What do you think ?
Kind regards

danjulio, 2022-12-08:
Gilles, I can't speak authoritatively about adding a RF amplifier but I guess it could be theoretically possible.  It looks like the ESP32 can be configured to control external RF switches which is what you'd need since you have to have both a TX path and a RX path.    You would use a GPIO to switch the TX or RX paths to the ESP32 antenna connection (I guess you could repurpose the GPIO I use and bring out on the Rev 4 board for the Hardware Serial IF). See the following link:

[https://docs.espressif.com/projects/esp-idf/en/v4.4-beta1/esp32s2/api-guides/wifi.html#wi-fi-multiple-antennas-configuration](https://docs.espressif.com/projects/esp-idf/en/v4.4-beta1/esp32s2/api-guides/wifi.html#wi-fi-multiple-antennas-configuration)

Of course you'd have also have high TX power at the other end since WiFi requires bi-direction communication and there could be legality or regulatory compliance issues and the like to consider.