danjulio/lepton #41: ESP 32 and Flir Lepton 3.5: OEM GPIO VSync Phase Delay

rmz-rider, 2023-01-23:
Hi, 

at first, thanks for all the support!
I am working on a project using ESP 32 and Flir Lepton 3.5. I am facing the problem, that there is no way to achieve negative phase delay. positive values 0 - 3 are working fine, but the negative ones or not accepted. Surely I am missing somthing, but what ?

Thanks a lot

Jonas

danjulio, 2023-01-23:
Hey Jonas,

I had looked at this but I don't have any experience with changing the Sync phase delay, sorry.  I assume you're using the Flir C SDK (IDD) library function LEP_SetOemGpioVsyncPhaseDelay?  Are you using my Arduino library?  I notice that the enum values for negative sync values are actually negative numbers.  Is it possible that there's a SW bug and the correct negative values aren't being written to the Lepton?

rmz-rider, 2023-02-14:
Hi,

thanks for your response. I am not using your library, but i also managed it without using Phase delay. 

however, thanks a lot