danjulio/lepton #43: shutter control through external pins?

leoshmu, 2023-10-20:
Hi, 
I have one of your tcam rev4 boards and it works amazingly well, thank you so much for putting this together. I am interested in a set up where the shutter is a bit separated from the housing, and this end up disconnecting the 2 pins of the shutter from the copper traces on the housing. I can easily solder wires to the shutter pins, and I'm wondering if the signal that controls the shutter to close is accessible on any of the pins that you have made available on the board. I see from the spec sheet that I_DDIO is a 310 mA current that is responsible for closing the shutter, but I can't tell what the housing traces connect to. 

Do any of the SLAVE IF pins on the board go high during FFC? Any advice you have about what signal on the board could be used to externally connect to the shutter (rather than rely on the two copper traces on the camera housing) would be much appreciated!

Best, 
Leo

danjulio, 2023-10-20:
Hi Leo,

I'd love to see how you're integrating an external shutter!  I never considered doing that so unfortunately I don't bring out any trigger for it.  Although you can send a command to initiate a FFC, normally my code just lets the Lepton manage that itself.

The Lepton does provide indication that it is about to and then performing a FFC in the telemetry data and my code reads the telemetry along with each image (see section 3.2 in the Lepton Engineering Datasheet that you can download from Group Gets if you don't have it already).

You could parse the telemetry data, probably in lep_task, and look at bits 4 & 5 of the Status Word (offset 3 & 4 of the lep_telemP data structure in the lep_buffer which you can get using the lepton_utilities lepton_get_tel_status function).  This will tell you when a FFC is imminent and/or in process and I think you'd set an external GPIO (you could use one of the Hardware IF GPIOs if you never ground Mode) during those states.  Although I'm not 100% sure about the timing.

One possible thing to consider is that the lepton uses the shutter temperature in its calculations during the FFC.  So either you need to assume your shutter is the same temperature as the Lepton's built-in temperature sensor or you'll need to do a more complex change to the code.  Flir suggests that you run the the module in manual mode with an external shutter and you manually command the FFCs (when you want and also when the camera requests it via the telemetry).  I think you'll have to provide the shutter temperature via the CCI (maybe the SYS FFC Mode Control command but not 100% sure).  The good thing about taking over FFC control completely would be you could assure the correct timing of an external trigger GPIO based on the mechanical timing attributes of your shutter with the actual FFC measurements performed by the Lepton.

Sorry I don't have an easier solution for you but I hope this points you in a fruitful direction.

Also, just FYI, I moved all things related to the tCam devices to a new github repo which is the one I currently update (https://github.com/danjulio/tCam).

Cheers, Dan

leoshmu, 2023-10-25:
Hi Dan,

Thanks so much for the feedback. I ended up taking a risk that worked - I
soldered directly to the leads on the housing that connect to the shutter
pins, then I can connect those wires to the pins of the shutter mechanism
with some careful soldering as well, and now the shutter can be positioned
wherever I like!

Will followup separately regarding why I am doing this!

Best,
Leo